sinara-hw/Thermostat_EEM #1: module specification

gkasprow, 2020-10-12:
@jordens 
- Do we need PoE?
- I assume that we can connect all AFEs to the same SPI interface, 4 chips per SPI
- DACs will share same SPI interfaces - two chips per SPI
- do we need individual PWMs for each TEC channel?
- I assume we want a downstream/upsteream EEM channel like in the Stabilizer. Was it tested already?
- what connectors? We can use ribbons with DSUB placed on separate panels as in the original Thermostat. But this takes a lot of space. We can also use HD DSUB and have 2 TECs and 4 sensors in each HDSUB15. In this way, noone will mix them with Thermostat or RS232 :) These days nobody is using VGA.

jordens, 2020-10-12:
> * Do we need PoE?

Not strictly need it if causes problems. But it fits the power envelope and would play nicely with the rest of the hardware.

> * I assume that we can connect all AFEs to the same SPI interface, 4 chips per SPI

Yes.

> * DACs will share same SPI interfaces - two chips per SPI

Yes.

> * do we need individual PWMs for each TEC channel?

You mean one MAX1968 per TEC? Yes.
Or do you mean the I/V limit PWMs? Those can be banked, let's say two TECs per set of voltages. But let's try to keep them as separate as possible.
But I might be misunderstanding you.

> * I assume we want a downstream/upsteream EEM channel like in the Stabilizer. Was it tested already?

With Stabilizer not that I know of. With Humpback and Urukul yes.
If there are space/pin problems, I'd rather drop the EEM connector but keep PoE.

> * what connectors? We can use ribbons with DSUB placed on separate panels as in the original Thermostat. But this takes a lot of space. We can also use HD DSUB and have 2 TECs and 4 sensors in each HDSUB15. In this way, noone will mix them with Thermostat or RS232 :) These days nobody is using VGA.

I'm only waiting for someone to connect a monitor to a Booster main board ;)

Do you mean [DE-15](https://en.wikipedia.org/wiki/D-subminiature)? Fine with me.
@j-hirsch and FBH should weigh in.

gkasprow, 2020-10-13:
The pinout could be as follows
[image]

gkasprow, 2020-10-21:
It looks like we will have individual PWM channels for all TEC driver settings.

gkasprow, 2020-10-28:
Initial placement
[image]

jordens, 2020-10-28:
When running one big tec from two channels in parallel, would we connect them on the header board?

gkasprow, 2020-10-28:
I'm not 100% sure if you can easily run two channels in parallel. These are synchronous converters - even a small differences in duty cycle will result in high cross-current flowing within the pair of shorted outputs. The header board is for debugging purposes.

jordens, 2020-10-29:
Aren't the drivers effectively current sources which should work in parallel? FBH indicated that they need one case with two in parallel. Best to disuss with @j-hirsch @schiemangk.

jordens, 2020-10-29:
Alternatively it might be easy to just use two electrically separate half-size TECs to get the same result.

hartytp, 2020-10-29:
looks nice! Any further thoughts about power management here? Can you get enough juice out of a single EEM for that many TEC drivers?

jordens, 2020-10-29:
It'll need active power management and load sharing anyway. EEM is probably not going to be the use case for us but the connector/cable is probably not going to be the limitation. It's just there because we can.

closed 2020-11-03 by gkasprow