sinara-hw/Thermostat_EEM #3: CPU signal assignment

gkasprow, 2020-10-28:
@jordens Please have a look at the CPU signal assignments.
It should be coherent with the STM32MX project
[Thermostat_8CH.zip](https://github.com/sinara-hw/Thermostat_EEM/files/5454705/Thermostat_8CH.zip)

jordens, 2020-10-29:
Looks good!

* What do you think about PWM for the fan?
* Can we add some minimal mounting studs or other arrangement for the fan?
* (Note to self: use AIN[0-3] vs AIN4 for NTC symmetry measurement, leakage/bad connection, thermocouple voltage check.)
* Is there something useful we can do with the ADC GPIO? Maybe enable/disable Vref+ (at R36, which I'd really like if possible)? Or at least a SMT LED?
* Agree to go for the 2 channel ADCs for compat and higher sample rate (I think).
* You are not worried that P3V3DVDD might make P3V3AVDD dirty. You change the power supply topology there I think.
* Can we use the remaining two pins on J2 for I2C? Preferrably a separate bus but if that doesn't work with BSS138 isolation.
* One EEPROM with EUI-48 is enough. IC12 can (DNP) go onto the header board (behind the J2 I2C: I2C2).
* What's the bit-bang SPI note for?

I'll go through the CPU assignments later.

jordens, 2020-10-29:
@gkasprow Could you elaborate a bit on the ideal diode design? All I know is [this](https://www.microfarad.de/blog/the-ideal-diode/). Doesn't the body diode always power downstream EEMs and also lead to POE_PWR_SRC being at best one body diode drop below 12V?

jordens, 2020-10-29:
Could we also add support for the unipolar MAX1969? Would only be a beefy jumper from LX2 to LX1 I think (and DNP L4, short C66 C67).

gkasprow, 2020-10-29:
the MOSFET is off by default. When forward currents starts passing via FET diode, the  from EEM to 12V rail, the T10A gets disable and MOSFET is on. Even cunducting FET has enough voltage drop to keep the T10 in off state.

jordens, 2020-10-29:
Ah. I just didn't look close enough. The body diode is reverse biased in the situation I was thinking of.

jordens, 2020-10-29:
@gkasprow I added a couple items to my list above. Really good.

jordens, 2020-10-29:
And finally one idea: can we add headers to support bridge/kelvin measurements on the ADC channels? Just a 16 way DNP IDC header (4 pins per channel with vexc+, sense+, sense-, vexc-) and two 0R jumpers added between vexc+ vsense+ and vexc-/vsense-.

jordens, 2020-10-29:
@gkasprow I added and clarified a couple of the points above.

gkasprow, 2020-10-31:
Let's add 8pin IDC connectors, it's easier to fanout them to DSUB using IDC connectors

gkasprow, 2020-10-31:
I moved the PWM_maxINeg1 to PB5 to free the PC7 which is now TIM8_OC2 and steers the fan PWM

gkasprow, 2020-10-31:
I don't want to make the excitation switchable using MOS switches. What I can accept is a miniature relay. Shall I go for it or 0R resistors are fine? You can always place jumpers on neighboring pins of the IDC8 connectors to enable 2-wire mode without a need for soldering 0R resistors.

gkasprow, 2020-10-31:
I rearranged the placement:
- the PoE module is oriented vertically to make conduction cooling easier. It does not obstruct the air flow,
- all drivers and inductors are arranged vertically so a single heatsink will be able to cover them
- DC/DC is assembled in-line with drivers so can be cooled with a single heatsink/fan
[image]

gkasprow, 2020-10-31:
@jordens do you want to run all channels with full power? 30W PoE would not be sufficient. We would also need a stronger DC/DC converter.

jordens, 2020-11-01:
ACK the timer move, the IDC connectors, and the arrangements

We'll do active power limiting for both temperature and total supply load. It would be nice to get a current shunt and CPU ADC channel on the 5V or 12V rail in addition to the per-channel power measurements that we already have.
If you already have nice 5V DC-DC blocks that can do a bit more current in mind, that would be interesting when using the Molex/barrel. But more PoE power is a bit awkward infrastructure-wise.

If you don't like to support alternating excitation with switches then let's leave it. Testpoints for GPIO01 instead please.

For the measurement configurations I'd like to ensure the following (I hope the notation is ok):

* Topology on the board should be: Vexc+ Re Vref+ Rr+ Vsense+, and then Vref+ Rs Vsense- Rr- Vref- Re Vexc-
* Rr+-, Rs are same size SMT reference resistors (like the current 5k). Re can be regular size (but having the same size as the 5k ref would be nice) and there only needs to be one set per ADC.
* Vexc+, Vref+, Vsense0+, Vsense0-, Vsense1+, Vsense1-, Vref-, Vexc- are on the IDC connectors.
* Vsense+, Vsense-, and the four Vref- (instead of the GND) are on the D-44 connector.
* For NTC/ratiometric (default): Re = 0, Rr+- = 5k, Rs = DNP, NTC between Vsense+-
* For 4/6 wire/kelvin/bridge: Re = custom/DNP, Rr+- = Rs = DNP, bridge betweeen Vref+- and Vsense+-
* For single-ended measurements with 4 NTCs per ADC (still fully ratiometric), Re = 0, Rr- = DNP, Rs = Rr+ = 11k, NTC_A between Vref- and Vsense+, NTC_B between Vref- and Vsense-.
* Make sure Vref+/- and ADC REF+/- are tied together well, and then tied to the low-passed AVDD/AVSS.
* Make sure AIN4 is tied to Vref-.

Forget about the I2C on the D-44 connector. I don't think that's worth it. Let's do identification and management on the current controler.

gkasprow, 2020-11-01:
@jordens The scenario could be simplified using one DPDT relay. Reed relays are recommended because they don't require minimum operating current as regular relays do. Can you make a paper drawing of the idea? I got lost with these Rr+- 
What do you mean by the topology on the board? PLacement of test points?
We don't have any free CPU pins, but I can add I2C current shunt.
stronger DC/DC is not a problem, I can also use two existing ones in parallel.

jordens, 2020-11-01:
Alternatively if you think it may be a good idea to I2C on the header board, then we could reduce the number of wires per TEC to 2. People seem to be talking about 5 A per contact and some 20 mOhm or thereabouts.

gkasprow, 2020-11-01:
What we can do is to reduce the number of TEC wires to two. We will spare 8 pins. Maybe, with SPDT relays we could scrap the internal IDC connectors ?

jordens, 2020-11-01:
Poor man's sketch of the four different use cases I have in mind and the IDC/D-44 and ADC connectivity.

[image]

Hope that's clearer.

I'd prefer to have at least the four Vref- on the D-44 connector. That would support 16 NTCs with still really good measurement quality.

jordens, 2020-11-01:
To support that kind of chopper mode, we'd need a DPDT relay per ADC to commutate Vexc+- between Vss/Vdd. The relay would be driven from GPIO0 by the ADC sequencer automatically (MUX_IO). It would work with all four use cases. The configurable ADC delay for that looks perfect.

jordens, 2020-11-01:
If we can put all four Vref+ and the four Vref- onto the D-44 connector (by scrapping the third TEC wire) then we can scrap the IDC connectors. Vexc+- and 6 wire mode isn't that interesting to me.
On the D-44 we'd have then 4*2 TEC+, 4*2 TEC-, 8 Vsense+, 8 Vsense-, 4 Vref+, 4 Vref-, SCL, SDA, 3V3_MP, GND. Exactly 44 contacts.

jordens, 2020-11-01:
A 3 m 22 AWG wire has 150 mOhm resistance. That with forward and return and two contacts, that drops 0.45 V at 3 A. Which is a bit annoying. 3 contacts would drop 0.3 V.
Then again, if drop becomes critical, we can just pigtail fatter cables from the D-44. The D-44 still looks like the best connector for the panel. The rest is a question of cabline.

gkasprow, 2020-11-01:
One can also use P3V3 and GND as vexc+-, probably using 3V LDO.

gkasprow, 2020-11-01:
I will use MIC45212 which can give off 14A of current.

gkasprow, 2020-11-01:
@jordens If we route also the GND and VDD, which is the case with I2C, then all possible variants could be possible. 
Since we don't need live reconfiguration, I would simply add DIP switches that configure the resistor network. 6 position DIP switch would do the job. Alternate actuation is also ok, we need only DPDT reed relay (20$). Reed relays also introduce very low thermoelectric potential.

gkasprow, 2020-11-01:
Such schematic should do the job.
[image]

gkasprow, 2020-11-01:
I'm not sure if we can use alternate excitation due to limitation of REF inputs
[image]
REF inputs can be buffered so we can use identical protection/filtering as in the case of analog inputs.

gkasprow, 2020-11-01:
This is the proposed protection of VREF inputs
[image]

jordens, 2020-11-02:
The REF+- wiring, MIC45212 look good.
I don't like the DIP switches for the resistor network configurations. I'd rather just have this done in rework.
With the Vexc commutation voltage offsets (due to thermoelectrics or MOSFET resistances) would not be a problem. They are cancelled by the measurement methods. But if -- as you say -- `REF+ >= 1 V + REF-` (I'm somewhat unsure whether this is required) then we can't do it this way. Let's leave it without Vexc commutation and just put test points onto GPIO0/1. I don't like the idea of placing the commutation between Vref and Vsense instead.
Also what's the lifetime cycles of those reed relays? We'd be switching them at 20-50 Hz. That's 1e6/day.

gkasprow, 2020-11-02:
1e6 is too much. They have 1e7..1e8 without load
Do you afraid of the DIP switch's performance? They are not hermetic and after some time the contacts will degrade. We can of course substitute them with reed relays but that would be costly. Another option is [OPTO-MOS](https://datasheet.octopart.com/AQY212EHAX-Panasonic-EW-datasheet-13536469.pdf) relays that have a few tens of Ohm ON resistance, 1e-12 off current, and no leakage to GND at all.

jordens, 2020-11-02:
My thought was to commutate Vexc between successive samples to zero out thermoelectric voltages etc. That can be implemented with the ADC sequencer at maybe 25 Hz. Vexc+ does not need to be very stable, not even between the two polarities, it already has 66 Ohm impedance in Thermostat-v2.0. Only REF+- need to track Vref+- accurately. But if we believe that REF+ > REF- then we can forget about it. Note that the manual also says "The device functions with a reference magnitude from 1 V to AVDD1" suggestiing that the magnitude of the differential counts. But then it says "REF− can span from AVSS to AVDD1 − 1 V" and "REF+ can span from AVSS + 1 V to AVDD1" which mostly exclude it. Let's consider actively commuting Vexc dead.
I don't like the DIP switches to implement the different resistor options in the signal path (or for commutating) because they are physically large, thus take up space and can have significant temperature gradients with thermoelectric potentials. And most importantly I'd expect that  they won't remove the need to change resistors for some applications and the sensor end of the cable will need reconfiguration and soldering anyway. If the single-ended ratiometric option and the bridge measurements prove useful and work well, then I'd be happy to reconsider the DIP switches. What do you think?

Also, if you go for a separate LDO for Vexc+, can we up the voltage (AVDD1 = Vexc+, and maybe also AVDD2) a bit towards the 5 V the ADC can handle? I'd expect dynamic range to increase a bit. I think we can share one LDO among all ADCs.

gkasprow, 2020-11-02:
that's not that bad with DIP switches area.
[image]
The idea was to just label them 1,2,3,4 where every number means one of the sensor scenarios. So the user simply toggles certain combinations of switches and gets desired input configuration.
I already implemented dedicated LDOs for analog and digital ADC rails.
5V AVDD is OK, the same for Vexc. The ADC can work with neraly any combination of IOVDD and AVDD
[image]
[image]
What if we toggle excitation of the input signals instead of REF?

jordens, 2020-11-02:
I still don't see us making much use of the DIP switches. If I needed DIP-switchable configurations and could live with them in the signal paths, I'd put them on the header board on other side of the cable.

Commutating Vsense+- doesn't help with thermoelectric voltages (you need to commutate the drive as well) and the ADC can already mux the inputs in any configuration. We'd need to commutate Vref+- while leaving the REF+- polarity fixed. That's tricky since that commutation would sit in the signal path where it easily introduces non-symmetric offsets.

jordens, 2020-11-02:
Do you think it makes sense to use two AD7172-4 instead of four AD7172-2? Price per chip is the same, programming and software is compatible. The only downside I see is the lack of the temperature sensor and internal reference (which we don't use).

jordens, 2020-11-02:
Could you explain the reset circuit with the diode between the SWD/JTAG NRST and the CPUT NRST/IC13 reset generator? The reset generator is push-pull but this way the JTAG/SWD can't reset the CPU (only monitor the reset?) and the connection seems useless then.

gkasprow, 2020-11-02:
True, I was already writing that since we have 3V3 and GND, we can realize all scenarios on the header board. So no resistors are needed. I would simply put only default configuration, rest of them can be build using dedicated header boards.
In case of ADC the price is not a big contribution, it's better to use components already used in other designs.

gkasprow, 2020-11-02:
the reset chip is push-pull, it keeps the RSTn pin low for a certain time after power on. Then its output is high and the debugger can pull RST low. The net name should be called RSTn, not RST
The JTAG JRST is not connected now.

jordens, 2020-11-02:
Ah. I mixed up where µC NRST is connected (RST and not CPU_RESETn).

You are right about the resistors not being needed at all here. They could and maybe should all be on the header board! But then we definitely need Vexc- (i.e. GND) and Vexc+ (P5V0A from the LDO, not 3V3) on the D44 connector.

* 4x2 TEC+, 4x2 TEC-
* 8 Vsense+, 8 Vsense-, very high impedance
* 1 Vref+, 1 Vref-, shared among the 4 ADCs, very high impedance
* 1 Vexc+ = 5V0A, 1 Vexc- (GND), low impedance
* SDA/SCL/GND/3V3_MP
* Use the remaining 4 contacts to beef up two TEC drives.

I'm still unsure whether it's better to have I2C on the header board or instead use those four contacts to beef up all TEC connections to three contacts.

jordens, 2020-11-02:
And it's not the best idea to connect AIN4 to REF- since REF- is available as a dedicated mux input already. If you don't have another idea, I think it's best to connect AIN4 to REFOUT (2.5 V reference that we don't use, which is not available as a mux input which gives us a nice (indirect) handle on measuring all relevant frontend voltages single-ended and absolute (measure REFOUT/REF- against REF+/REF-, then measure AIN0-3/REF- against REF+/REF-.

gkasprow, 2020-11-02:
why not use 3V3 LDO on the header board and supply I2C from Vexc+? I2C won't be used often. In this way we would keep all 4 vref pairs

jordens, 2020-11-02:
Yes. Sounds good.

gkasprow, 2020-11-02:
So, to conclude, I will keep only the default symmetrical ratiometric connection.
[image]

jordens, 2020-11-02:
Yes. Exactly like that. Connect Vref- to REF- and Vref+ to REF+. Bring VREF_5V0 and GND to the connector. No need for the net-ties NT6 and NT2 anymore.

gkasprow, 2020-11-02:
Right, the net ties were needed to avoid the wrong GND routing

jordens, 2020-11-02:
I'd like to change the TEC_U_Meas circuit to be referenced to the MAX Vref 1.5V (not (CPU ADC_REF + REF_GND)/2). That makes these MAX voltages all consistently referenced.
Could you also remove the DAC_FB circuitry? I don't want to use it and instead find the overall offset with ITEC or TEC_U_Meas vs the MAX Vref 1.5V. At least DNP R69 by default and feel free to use those CPU ADC inputs for something else (sensing the 12V current shunt for example).

jordens, 2020-11-02:
I'd also like to be able to measure the P3V3 voltage because the PWMs are proportional to it. The best way I see is to connect VBAT to a scaled P3V3 and then measure VBAT relative to VREFINT (~1.2V) or P5V0A/CPU_ADC_REF.

gkasprow, 2020-11-02:
Right, we have ITEC and TEC_U so these are suffienct means to verify that DAC works. We have free ADC inputs, so we can measure 3V3 directly.

gkasprow, 2020-11-02:
[image]

gkasprow, 2020-11-02:
the 3V CPU reference generation uses 1% resistors and opamp buffer with offset of 7mV max . Do we need better precision?

gkasprow, 2020-11-02:
I mean this circuit 
[image]

jordens, 2020-11-02:
No. 1% is fine. for me.

gkasprow, 2020-11-02:
I added a new schematic release. Please check it.

jordens, 2020-11-03:
Thanks. I filed individual issues and I think we can close this one.

closed 2020-11-03 by gkasprow