sinara-hw/Thermostat_EEM #11: MAC1969 options

jordens, 2020-11-03:
[image]

We should move the jumper behind the inductors L2 and L4 to utilize them in parallel and C67/C66 as well.

This should be:

* a jumper between the right sides of L4 and L2 (DNP for bipolar)
* a jumper between the right side of L4 and OS2 (DNP for unipolar)
* a jumper from OS2 to GND (DNP for bipolar).

Can we easily make the jumper and the CMC capable of 6 A.

gkasprow, 2020-11-04:
I replaced the CMC with a[ 1.8uH 9A](https://pl.mouser.com/datasheet/2/336/P850-1527002.pdf) one. Slightly bigger but fits nicely.

gkasprow, 2020-11-04:
I used dedicated 1206 0.5mR jumpers. Standard 0R 1206 resistors are roughly 50mOhm

gkasprow, 2020-11-04:
[image]

closed 2020-11-04 by gkasprow