sinara-hw/Thermostat_EEM #18: Allow Molex/Barrel 12V to be measured with the shunt

jordens, 2020-11-03:
These should drive P12V0_R

[image]

jordens, 2020-11-03:
And the EEM (by default) as well.

[image]

gkasprow, 2020-11-03:
[image]
[image]
[image]

The only issue is that the load connected to EEM is not measured.
What we can do is to add yet another FET that would redirect the current when powering downstream port
[image]

jordens, 2020-11-04:
I don't think that's necessary. But I'll leave it to you.

gkasprow, 2020-11-04:
let's keep it simple then.

closed 2020-11-04 by gkasprow