sinara-hw/Thermostat_EEM #19: I2C isolation

jordens, 2020-11-03:
Let's use CPU_RESETn to drive those FETs. It's later than P5V0 and later than P3V3

[image]

jordens, 2020-11-03:
I'd also place the LM75 on the EEM side of the I2C bus (I2C_SDA/SCL instead of I2C4_SDA/SCL)

[image]

jordens, 2020-11-03:
And we need 10k pullups to P3V3_MP on the EEM side of the I2C bus (I2C_SDA/SCL).

gkasprow, 2020-11-03:
I'd use P5V0A which wakes up as the last one after P3V3. It will result with lower Rdson.

gkasprow, 2020-11-03:
[image]

closed 2020-11-03 by gkasprow