sinara-hw/Thermostat_EEM #22: PoE again

jordens, 2020-11-03:
A few thoughts:

Looking at the [SMSC app note on PoE](http://ww1.microchip.com/downloads/en/AppNotes/en562768.pdf), they recommend a different circuit for protection: back-to-back 5V TVS/Zeners accross RD and TD. Apparently lots of capacitance allowed here (50 pF differential).

Second, I'm a bit worried about the mismatch between the TVS datasheet and the schematic symbol:

[image]
[image]

Unlike the symbol, in the datasheet there is no explicit protection of the data lines to AVDDT_PHY at all. It's only a Zener + forward diode to GND. And that Zener has 8 V clamping voltage: higher than the abs max of the PHY, higher than expected from the schematic symbol if it was only a forward diode drop to 3.3V (AVDDT_PHY), and higher than the 3.5 V "punch-through" + forward diode drop of the SMSC design.

Third, I think in our design the AVDDT_PHY protection of 8V is still a bit loose compared to what the PHY can stand (yes there is plenty of capacitance to dampen the surge but still).

Also re https://github.com/sinara-hw/Thermostat/issues/83

gkasprow, 2020-11-03:
True, the symbol is wrong!
This could explain why PHY died.
Good catch!
We also need to replace it with the Stabilizer module.
Fortunately, I did not update all other designs with this part!
What we need is not necessarily Zener diode but fast protection diodes where excessive charge can be dumped to the supply rail. This would work
https://datasheet.octopart.com/PRTR5V0U4Y%2C125-Nexperia-datasheet-87582156.pdf

gkasprow, 2020-11-04:
I'm not sure if the cathodes can be connected to 3V3 rail. It will limit the max voltage produced by the PHY chip (there is a transformer with center tap). Maybe we should connect it to 5V rail instead...

gkasprow, 2020-11-04:
the maximum ratings say:
[image]

gkasprow, 2020-11-04:
on the other hand, the TX won't exceed 1V
[image]

gkasprow, 2020-11-04:
which means that we have AVDDPHY + 0.5V on one output and AVDDPHY -0.5V on complementary output. The TVS diodes should not clip such amplitude

jordens, 2020-11-04:
I always thought they were complementary and just switch between 0 and avddphy. I would not expect them to get above avddphy actively. There will always be one inactive half of the winding. Induced voltages in the inactive half should be clamped to avddphy+diode in the phy. No?

gkasprow, 2020-11-04:
These are analog signals. 3-state sequence. The transformer central tap is connected to 3V3, so when one output pulls low, the other gets the same value above the bias. The PHY does not have the clamping diode (it accepts up to 5.3V), it's like an open-drain circuit. That's why it can be vulnerable to damage.

closed 2020-11-04 by gkasprow

jordens, 2021-05-04:
Note for the next time we revisit this: We could also try "more robust", "industrial" PHYs like the DP83822.

gkasprow, 2021-05-04:
Good ida. But I would do it in case we observe chip failures.