sinara-hw/Thermostat_EEM #4: P5V0A as Vexc on the header board

jordens, 2020-11-03:
[image]

Not P3V3_AFE.

jordens, 2020-11-03:
And I think we may want to add a polyfuse on Thermostat towards the D44 connector P5V0A.

gkasprow, 2020-11-03:
This comes from 300mA LDO. I'm not sure if it's worth adding PTC to BOM

jordens, 2020-11-03:
Right. I guess the LDO will just shut down.

gkasprow, 2020-11-03:
yes, it will get thermal shutdown.

gkasprow, 2020-11-03:
on the header board, we have 33Ohm series resistor already

closed 2020-11-03 by gkasprow

jordens, 2020-11-04:
That's for the P3V3_AFE LDO, right? For the excitation voltage to the sensor connectors I'd like P5V0A with no large added resistance, just some filtering. (e.g. load cells are just a couple hundred Ohm only and like to pull > 10 mA at 5 V.)