sinara-hw/Thermostat_EEM #7: P5V0REF

jordens, 2020-11-03:
I don't think wen need P5V0REF.

* And the DACs also don't need to be that accurate. They are in-loop and relative to the somewhat unstable MAX Vref.
* The CPU ADC reference is as inaccurate as the resistors/buffer.
* The P5V0A is clean and has enough power

[image]

jordens, 2020-11-03:
Maybe it's much better to have a 3V reference for the DACs and the CPU ADC. Then there would be no need for the IC10 buffer, the R54/R70 divider and the R88/R87 divider. Drive IC5 VDD from P5V0A with a bit of resistance to isolate it.

gkasprow, 2020-11-04:
Good idea. I don't remember what was the rationale behind supplying the DAC REF from 5V. The DAC performance is not critical here.

closed 2020-11-04 by gkasprow