sinara-hw/Thermostat_EEM #23: D44 assembly

jordens, 2020-11-07:
With the connectors like this, the two boards have to be separated before soldering the D44 connectors. Breaking on the drill line after soldering probably won't work because you can't bend the connectors when connected. Is this the intention?

[image]

gkasprow, 2020-11-07:
breaking will be easy
[image]

The connectors will be mounted after all, but U are right, they can be also mounted automatically - we have such machine. I will separate them a bit.

jordens, 2020-11-07:
Yes. I have seen the break line. But that breaks by bending and you can't bend the connectors when mated. So you'd have to make sure that they are mounted after breaking. I don't know how the assembly works and how much is done manually, but it might be easier to allow the board to be broken off after all assembly. That would require separating the boards by more.

gkasprow, 2020-11-07:
Actually we have a special machine that separates the boards. We don't break them manually.
It's logistically easier to do after automatic soldering. Once you separate them, It's hard to grip it for automatic soldering...

jordens, 2020-11-07:
Right. As mentioned I thought that you solder before separating. Then I wonder how the machine separates without bending or shearing (which would both stress the mated connectors a lot) and also without direct access to both sides of the drill lines.

jordens, 2020-11-08:
I guess if it's a saw that does the separation, this is fine.

gkasprow, 2021-01-19:
I separated the two boards
[image]

closed 2021-01-19 by gkasprow