sinara-hw/Thermostat_EEM #29: polarity marker for jtag connector

jordens, 2020-11-08:
* Since it isn't keyed, it should at least get a pin-1 marker, https://github.com/sinara-hw/Thermostat/issues/76
* Also make sure that the connect fits, see https://github.com/sinara-hw/Thermostat/issues/75

[image]

gkasprow, 2020-11-08:
The PN used in the design is EOL.
What about using this one from Samtec, with the key?
[image]

jordens, 2020-11-09:
The non-keyed connectors would fit that one as well, right?
Then that's good. Still, please add a pin-1 marker if the footprint doesn't have one.

gkasprow, 2020-11-17:
Harwin M50-3600542R fits well existing footprint

gkasprow, 2020-11-17:
Connfly ZL322-2X5P works as well

gkasprow, 2021-01-19:
fiexd. I used  M50-3600542
[image]