sinara-hw/Thermostat_EEM #30: AD717[2567]

jordens, 2020-11-09:
Quick analysis of pin- and software-compatible alternative ADCs.

tl;dr: The AD7175 is an interesting option over the AD7172 for fast loops (faster than 50 Hz data rate).

* 2,5,6,7 are all pin-compatible and register compatible. This is stressend by the common ADI noOS driver.
* AD7175 is very similar to the 2, (almost) same base rms noise, but higher base sample rate (250 kS/s vs 31 kS/s, factor 8).
* AD7175 has 8 times higher clock/conversion frequency and about 8 times the current consumption./
* AD7176 is the AD7175 without buffers
* AD7177 has 32 bits but the same in-band noise as the AD7175. It's useful if the data rate is under 400 Hz compared to the 5.
* The 5's higher base sample rate (before the filters) at the same base quantization noise directly leads to less voltage noise (about sqrt(8)) at the same output sample rate.
* Once the filtered voltage noise drops below the filtered (24 bit) quantization, the later dominates. That's at 400 Hz and 60 Hz for the 5 and 2 respectively.
* Tempco is dominated by the 5ppm/K reference resistors. There are 2ppm/K resistors available from the same series. Going to <1 ppm/K reference resistors (through hole) and six wire measurement (remote excitation) or a <0.5 ppm/K current source and four wire measurement would start to make the ADC the dominant part. Both alternative setups are possible with the current design.
* The ADCs have 0.2 (AD7172) and 0.4 ppm/K (AD7175) typ. But looking at the actual distribution of those drifts they are indistinguishable to me. Unclear what the measurement accuracy is and the device drift might even be indistinguishable from zero.
* 10k||(5k+5k) Johnson noise in 500 Hz is 0.2 µV. 24 bit quantization is 0.17 µV rms. The two reference resistors additionally contribute 0.06 µV/V current noise (in one decade bandwidth, 1/f) each. We run them at 1.25 V.
* NTCs also have current noise. Maybe even 1 µV/V (20 µK/V) in 0.01-1 Hz depending on the design ([Ryger et al. Rev. Sci. Instrum. 88, 024707 (2017)](https://pdfs.semanticscholar.org/1198/54a2bf7d9f0db15f1273141bfb6e8449f83b.pdf). We run at 2.5 V.
* 50 Hz should be the sweet spot for the AD7172 (ENOB 24 bit and 50 Hz notch with minimal latency), 400 Hz for the AD7175
* The current design has a 160 Hz RC filter (200n, 10k||(5k+5k)) before the ADC limiting the bandwidth and latency
* Thermal gradients between effective temperature of interest and sensor are typically larger than 1 mK (50 µV, 1 µHz-0.1 Hz, after settling, with a decently designed multi-layer actively shielded commercial optical ULE cavity).
* For faster loops with closely coupled TECs, thermal mass, and sensors, or other types of sensors and measurements, the lower noise of the AD7175 is superior.

gkasprow, 2020-11-09:
For me is fine, it would be good to use it in both Thermostat designs.

jordens, 2020-11-09:
Agreed. These were just my notes what would need to be done to get to faster and even more accurate/stable/low-noise control loops.

closed 2020-11-09 by jordens