sinara-hw/Thermostat_EEM #34: J7,8,11,13 Vrefbx- close to GND

jordens, 2020-11-10:
Please move Vrefbx- to pin2 close to the GND pin1 on J7/J8/J11/J13.

The pinout from pins 1-8 should be GND, Vrefbx-, SENSBx_2, SENSBx_1, SENSBx_2, SENSBx_1, Vrefbx+, P5V0_AFE.

[image]

gkasprow, 2021-01-19:
done 
[image]

closed 2021-01-19 by gkasprow

jordens, 2021-01-19:
Good idea! But then R129 and R130 should be treated the same. They should also move to the header board. and be placed accross pins 8-7 and 1-2 respectively.

And I'm not sure we need this section then anymore:
[image]

gkasprow, 2021-01-19:
right!

gkasprow, 2021-01-19:
done 
[image]