sinara-hw/Thermostat_EEM #35: 0805 footprints for reference resistors

jordens, 2020-11-10:
Currently the reference resistors dominate the temperature coefficient (an order of magnitude more than ADC gain tempco, lead resistance tempco, lead thermoelectric emf, ADC offset tempco, resistor voltage coefficients).
A fix would be to get better resistors.

* Those could in principle be on the (actual) header board. That would also cancel lead resistance tempco and lead resistance offsets/instabilities.
* Alternatively we could go to 0.2 ppm/C resistors. 5k is available in 0603 (Y16365K00000F0W).
* But many more resistors are available in 0805 size from 0.2 ppm/C (Y16295K00000T9R) to the existing 5 ppm/C.

Could we go for 0805 size reference resistors (e.g. TNPU08054K99BZEN00 for R91 R44 R132 R50)? Resistance can be anywhere from 4.7-5.2k.

gkasprow, 2020-11-10:
Sure, let's go.
The most precise ones are in the miniMELF package (1206). We have a lot of space.

jordens, 2020-11-11:
Are you sure? On digikey, the common SMT resistors around ~5k and 0.2 ppm/K are 0805, some 1206, no miniMELF.

gkasprow, 2020-11-11:
It used to be the case 15 years ago. It was easier to seal miniMELF. Bit it seems the situation changed recently :)

gkasprow, 2021-01-19:
Let's go for Y16245K00000T9R. I have them already in my libs.

jordens, 2021-01-19:
Agreed

gkasprow, 2021-01-19:
I added them on the header board, the ones close to DACs will be DNP
[image]

closed 2021-01-19 by gkasprow