sinara-hw/Thermostat_EEM #39: support for cryogenic temperature sensors

gkasprow, 2021-01-20:
I want to use Thermostat to sense 4K enviorenment using silicon sensors, for example one of [these](https://www.lakeshore.com/products/categories/overview/temperature-products/cryogenic-temperature-sensors/dt-670-silicon-diodes)
The diode characteristics is below
[image]
it should be easy - connect the ADC references to external reference voltage, for example 3ppm LM4140 which gives 2.048V
and then use current source based on opamp + LM4140 (1.024V) to generate 10uA current.
[image]

 Then use 4 wires to connect the sensor.
@jordens do you think that makes sense?

gkasprow, 2021-01-20:
Another option is to use a current sink that would use a common reference source for all channels, then eight opamps + 8 low leakage current JFETs

jordens, 2021-01-20:
The ADC has an internal 2.5 V 2ppm/K reference already. That's usually enough. I'd build a current source on the header board and implement Kelvin sensing.

Beyond that:
If you can make a good star-ground near the diodes and sense that with one of the input channels you'd get more diodes per ADC. You can also put several diodes in series of a CC source.
Maybe you could even use the REF+, REF- terminals as two additional independent inputs if you use the internal reference. The datasheet seems to indicate that they can be used as such. With a good star ground and REF- connected sensing, you could measure four diodes (AIN[0:3] against REF-).
Calibration, uncompensated Thermocouple voltages and lead resistance for non-Kelvin are probably the biggest problems.

gkasprow, 2021-01-20:
Sure, by writing "Then use 4 wires to connect the sensor." I meant Kelvin connection. I was thinking about using two similar references or even one with the hope that their tempco will compensate to some extent :)
2.048V reference will let me use more ADC range but the ADC one has a lower tempco.
I ordered calibrated thermometer so can use it as a reference. I want to implement some interpolation in the Thermostat firmware, connect 8 sensors attached to the same cold head as reference sensors and do the calibration automatically. It won't be that simple at 4K due to several possible issues, but I won't know until I try.

gkasprow, 2021-01-20:
Here is the proposed AFE for cryo sensors. The sensor wires are long, I want true differential inputs. 8 channels are fine.
[image]

gkasprow, 2021-01-20:
I'm not sure if the ADC will handle 5V common-mode signal..

gkasprow, 2021-01-20:
I can always add 120k resistor in series with P5V0_AFE...

jordens, 2021-01-21:
* With Kelvin I only wanted to say that I think your approach of 4-wire is exactly right.
* Why do you need the JFET and low-side drive of the diode? Can't the opamp drive high side excitation of the diode (if you want, through a resistor to protect the diode a bit)? (8: opamp out through 120k protection, 7+6: sense, 5: 120k burden/shunt to GND). That would also get the sense signals common mode away from the rails.
* I don't think a second reference for the ADC will help.

gkasprow, 2021-05-21:
right! we can scrap the MOSFET this way

closed 2021-05-21 by gkasprow