sinara-hw/Thermostat_EEM #42: High power PoE

gkasprow, 2021-02-15:
There are 60 and 85W PoE [modules](https://silvertel.com/ag5800/) avaialble from Silvertel.
They need an 802.3bt class switch but are backward compatible with AT sources

jordens, 2021-02-15:
Nice. To me AT compatibility is crucial, I don't see BT switches very often.
I'm really looking forward to active temperature, total power management, sharing, and dynamic limiting on Thermostat and especially Thermostat_EEM.

gkasprow, 2021-03-01:
the  802.3bt  module has 17mm, it's too high for 4HP EEM

gkasprow, 2021-03-01:
But, [AG5810](https://silvertel.com/ag5810/) would fit. But would require significant board re-design

jordens, 2021-05-04:
Isn't the fan mezzanine (incl screw heads etc) too high for 4 HP anyway? Even if it just fits, wouldn't the fan choke on the proximity to the adjacent board?

gkasprow, 2021-05-04:
The idea is to use it when a convection cooling or fan tray is not used. In such cases, we could either install a 2HP filler or go for an 8HP panel.