sinara-hw/Thermostat_EEM #43: Stm32h7

jordens, 2021-02-17:
Let's consider moving to the stm32h7 right away. Price is almost the same, more power, newer architecture, lots of experience with it on stabilizer now.
I think there is a pin compatible stm32h743 to the f4 that we are using. It might just be a bom.change.

@gkasprow what do you think?

gkasprow, 2021-02-17:
Let's do whatever is easier from SW's point of view.

jordens, 2021-02-18:
Ok. Let's do stm32h743zit6 as on Stabilizer then. The H7 teething problems appear to be over and I can see the need for some additional computation power here.
I think this is a drop-in replacement but it's probably worth verifying the complete pin config.

c.f. old discussion at https://github.com/sinara-hw/Thermostat/issues/12

gkasprow, 2021-03-14:
done, please check the CubeMX project
[Thermostat__EEM_8CH.zip](https://github.com/sinara-hw/Thermostat_EEM/files/6137141/Thermostat__EEM_8CH.zip)

SingularitySurfer, 2021-03-17:
Hi, I looked at the schematic and I couldn't find anything wrong with the pinout. But I also don't know the exact timer/ADC configuration etc. and generally this is the first time I properly looked at Thermostat in depth. So maybe someone else should check as well.

jordens, 2021-03-18:
@gkasprow Is there a chance to group the timer channels a bit clearer? Specifically:

* Can we put PWM_maxINeg1 onto the same timer (TIM8) as the three other? I.e. swap FAN_PWM and PWM_maxINeg1.

gkasprow, 2021-05-21:
they are not mounted by default
[image]

closed 2021-05-21 by gkasprow