sinara-hw/Thermostat_EEM #48: Stock problems v1.0

filipswit, 2021-11-24:
Due to stock problems: 

- Alternate power supply section has been added. 
[image]

- Indicator leds are 4x green instead of 2x green and 2x red
- ARJM11C7-114-BA-EW2 changed to 2337992-8

jordens, 2021-11-24:
That's a 6 A limit now. A bit small compared to the 14 A of the MIC. It would be good if we can at least make 8 A, better 12 A (3 A per TEC not including losses). We have a high power 12 V connection that can supply enough. I'd rather not be limited by the 5 V power supply.

filipswit, 2021-11-24:
It's hard to find something what is small enough, high current and it's possible to buy in this year.

jordens, 2021-11-24:
If this is an emergency alternative, OK. Can we make the boards so that they can be upgraded to the MIC by rework when available?

akaminska, 2021-11-24:
@jordens this is only a temporary solution for the purpose of the prototype series, in order to be able to manufacture it now. If you think that it will be really useful, we can try to squeeze in one more 584-LTM4623EV#PBF

filipswit, 2021-11-24:
@jordens We treat LTM4623 as temporary solution. MIC section is left as it was, we would like to mount all RC components and as soon as MIC will be available, upgrade will be possible. 

[image]

filipswit, 2021-11-25:
@jordens finally i squeezed there 3rd DC/DC so now we have 3x3A.

closed 2021-11-25 by filipswit