sinara-hw/Thermostat_EEM #51: Ethernet PHY can only negotiate 10BASE-T link

SingularitySurfer, 2022-02-03:
The lan8742 PHY is not able to negotiate a 100BASE-TX connection with the other side of the Link. 

This was tried with 2 different ~1m RJ45 cables (One is Cat.5E) and 3 different USB-Ehernet adapters. With one of the adapters it does not work at all, with the other two the link status glitches between on and off for about 10 sec until the link stabilizes to on with a 10BASE-T connection (which is determined by reading the PHY status regs).
The cables don't make a difference and all adapters and cables work instantly with Stabilizer.

The only difference to Stabilizer in the schematic are these 33 ohm inline resistors:
[image]

In the layout the traces are much longer.

gkasprow, 2022-02-03:
The thermostat (2 channel version) is using the same resistors. I didn't check what speed it negotiates though.

SingularitySurfer, 2022-02-03:
Just checked, it negotiates 100Mbit and also instantly connects.

The problem is that the drivers we are using only support 100BASE-TX full-duplex..

jordens, 2022-02-03:
@SingularitySurfer quickly check that the ESD array footprint, schematic, and datasheet actually match. There were problems with that before.

gkasprow, 2022-02-03:
which revision of the Thermostat do you have? these resistors were added quite recently.

SingularitySurfer, 2022-02-03:
I've got Thermostat v2.0 and Thermostat-EEM v1.0.

ESD array footprint, schematic, and datasheet are the same for both Thermostats and Stabilizer.

SingularitySurfer, 2022-02-03:
The resistors are not present in Thermostat v2.0 I think.

gkasprow, 2022-02-03:
right. They were added in v2.1

SingularitySurfer, 2022-02-03:
Unfortunately I think I ESD-damaged my PHY while bridging the resistors. I'll try swapping the PHY and bridging the resistors tomorrow.

jordens, 2022-02-03:
@gkasprow where does the 33R series come from, was it a guess to add some more ESD robustness or is that a proven design?
The [phy schematic checklist](https://ww1.microchip.com/downloads/en/DeviceDoc/LAN8742A%20QFN%20Rev%20A%20Schematic%20Checklist.pdf) doesn't have them. But it also doesn't have the ESD diodes so it's of limited value.

gkasprow, 2022-02-03:
It was recommended in PoE app note to prevent PHY from dying

jordens, 2022-02-03:
The one you referenced [here](https://github.com/sinara-hw/Thermostat/issues/83#issuecomment-719920167) talks about 0-5 R.
But for that short cable that @SingularitySurfer I wouldn't expect it to make a difference.

gkasprow, 2022-02-03:
you can short them and see if that makes any difference. So far nobody complained about Thermostat...

SingularitySurfer, 2022-02-04:
Turns out I ESD damaged the ESD protection diode array such that there appeared another diode to from the data lines to ground..

SingularitySurfer, 2022-02-04:
With a new PHY, no ESD protection diode array and shorted resistors I instantly get a 100BASE-TX connection.

gkasprow, 2022-02-04:
the ESD diodes must be there. Otherwise after plugging PoE you will damage PHY instantly,

gkasprow, 2022-02-04:
you can "borrow" the ESD array from the I2C circuit close to EEM connector (D17)

SingularitySurfer, 2022-02-04:
Since I don't use PoE I don't need the diode array right now. Would you like me to check if the diode array in there makes a difference? I don't think that's so likely since it's also there on other devices where it works.

gkasprow, 2022-02-04:
these TVS arrays have very low capacitance and are used in Stabilizer already.
The only thing that changed are these resistors. So I think they caused the problem.
Can you solder 5R1 and see if it works with a long cable?
The traces don't matter, they are still relatively short.

SingularitySurfer, 2022-02-04:
I neither have resistors that small in physical size, nor any resistor < 10 ohms here..

SingularitySurfer, 2022-02-04:
Oh and actually since Thermostat has a different MCU it could be that the drivers there support 10Mbit ethernet.

gkasprow, 2022-02-04:
Isn't the RUST stack the same?

SingularitySurfer, 2022-02-04:
No, it uses a different HAL and ethernet driver. Unf I can't find it in the m-labs code right now but [here](https://github.com/stm32-rs/stm32-eth/blob/f246fae1dd37d3d20b07e84b4baaa71826388139/examples/arp.rs#L178-L185) they use the same driver a the m-labs code uses to look for any connection in the link, incl. 10Mbit. Not sure if it would work with 10Mbit after that tho.

gkasprow, 2022-02-04:
OK, that's why nobody noticed the difference

SingularitySurfer, 2022-02-04:
@jordens We don't have a new Thermostat here right?

jordens, 2022-02-04:
Not newer than what you have been playing with.

SingularitySurfer, 2022-02-09:
Some more remarks on the ESD array:
- Between Stabilizer and Thermostat-EEM I have noticed that the schematic symbol, BOM part and footprint name are exactly the same. 
- The actual physical footprint seems to be slightly different with a bigger clearance between solder mask and pad on Thermostat.
- The diode array place on the actual board is not [the one in the BOM](https://octopart.com/datasheet/drtr5v0u4s-7-diodes+inc.-30954912). The marking is TG5, whereas it should be TG1 according to the BOM part. Other ESD arrays on Thermostat have the same TG5 marking.  A Stabilizer v1.2 I could look at has the TG1 marking. Unf. I was not able to find the corresponding datasheet for the TG5 marking, since I cannot search for the marking and the datasheets I could find had other markings.

It is therefore still possible that the diode array makes a difference.

jordens, 2022-02-09:
https://www.mouser.com/datasheet/2/115/DRTR5V0U4SL-476981.pdf Note the completely different pinout.

jordens, 2022-02-09:
@SingularitySurfer If this pans out, take home message for you is [check things](https://github.com/sinara-hw/Thermostat_EEM/issues/51#issuecomment-1029065804) first.

gkasprow, 2022-02-09:
There should be mounted DRTR5V0U4S-7 which has the correct layout. Essential is that pin 5 must have all cathodes connected to it.

jordens, 2022-02-09:
@gkasprow According to the marking it would be DRTR5V0U4S**L** with the different pinout. Maybe a chance to visit manufacturing and straighten them out ;)

gkasprow, 2022-02-09:
I wrote them an email. Hope they didn't assemble it in other modules.

jordens, 2022-02-09:
Anyway, no harm done. Then also no indication of ESD damage, just "undesired diodes". Good that we figured it out. Thanks all!

gkasprow, 2022-02-09:
OK. Production people assembled DRTR5V0U4SL-7 because one of the engineers told them that it's the same part.

jordens, 2022-02-09:
All the way to the name of the engineer! ;)

gkasprow, 2022-02-09:
The other TVS are also bad: D3, D4, D27, D35, D36

jordens, 2022-02-09:
Maybe also on other Thermostat and Stabilizer builds.

SingularitySurfer, 2022-02-16:
Hi @gkasprow, we have noticed that diode arrays D21 (USB) and D26 (JTAG) are different arrays with the marking "TV8" and obviously seem to work as intended in case of the JTAG. In the BOM they are the same part as the affected diodes. How come these two are different?

jordens, 2022-02-16:
Ah. USB may be 5V, too close to the Zener voltage of the TVS.
Maybe JTAG can/could do higher voltage as well. Hence those should be different parts than the others?

filipswit, 2022-02-16:
To clarify, this are mounted diodes:
D3, D4, D27, D35, D36 - DRTR5V0U4SL-7
D21, D26 - D5V0F4U6S-7

jordens, 2022-03-31:
Production issue, not a design issue. Nothing to be done.

closed 2022-03-31 by jordens