sinara-hw/Thermostat_EEM #52: Future Thermostat-EEM design whishlist

SingularitySurfer, 2022-02-16:
This a list of design changes that we could consider in the future. To be discussed. Some items might not make sense in practice or due to other constraints. 

- If a laser module is rack-mounted together with Thermostat-EEM, it would be nice to have a way of internally connecting the laser module with Thermostat. Currently the board only features front-facing connectors. Probably a connection with ribbon-cable/EEM-connector would make sense here? Also the top side component density is relatively high already.

jordens, 2022-02-16:
Best to do one issue per item. That focuses the discussion.

SingularitySurfer, 2022-02-16:
Ok, I'll rename the issue etc.