sinara-hw/Thermostat_EEM #55: TEC Driver REF pin loading and capacitance

SingularitySurfer, 2022-02-21:
The TEC Driver IC has a Reference pin that outputs the internal reference voltage. This pin should have a 1uF capacitor according to the datasheet, Thermostat curretly has a 10nF. 
The pin can sink 1mA but only source 10uA. With the current loading the pin is pulled from nominal 1.5V to ~1.05V.
[image]

jordens, 2022-02-22:
This was a bug from https://github.com/sinara-hw/Thermostat_EEM/issues/46
I meant adding an 10k R 10n C filter to VREF, after the 1u C and not replacing the 1u with a 10n C and adding an R 10k. That doesn't work.

jordens, 2022-02-22:
Still, there seems to be something loading it. @SingularitySurfer sure you measured the 1.05V on the pin 4 side or R224?
In any case the voltage divider at IC14 is broken due to the non-zero impedance of R224.

SingularitySurfer, 2022-02-22:
> Still, there seems to be something loading it. @SingularitySurfer sure you measured the 1.05V on the pin 4 side or R224? In any case the voltage divider at IC14 is broken due to the non-zero impedance of R224.

Actually I measured on the wrong side of the resistor, sorry! The highlighted net on the pin has 1.5V. Behind R224 there is 1.05V left.
[image]

jordens, 2022-02-22:
Ok. Then it's just the divider. And the RC. Should fix both.

gkasprow, 2023-03-06:
R224=0R, C168=1uF

jordens, 2023-03-06:
@gkasprow I think that's too much load. VREF can do +10µA to -1mA. We are loading it with 25k from 4.3V worst case (see IC14).

gkasprow, 2023-03-06:
Right. So let's add another opamp to buffer it. It can be one for all four channels

gkasprow, 2023-03-06:
Or, let's replace IC14 with LM358

gkasprow, 2023-03-06:
[image]

jordens, 2023-03-07:
The buffer should also drive the other vref users, e.g. to the CPU.

gkasprow, 2023-03-07:
Fixed in v1.1rc2

closed 2023-03-08 by gkasprow