sinara-hw/Thermostat_EEM #56: #SHDN and FREQ signals need voltage level translator

maciejprzybysz, 2022-02-21:
Same issue as in Thermostat standalone version, details described here:
[Thermostat Issue #110](https://github.com/sinara-hw/Thermostat/issues/110)

SingularitySurfer, 2022-02-22:
Thermostat-EEM has the 10k pulldown, 2k to MCU resistors on the #SHDN pin, leading to 2.7V at the pin.

> The same issue applies to all previous versions, but despite the difference in voltages, it has worked fine so far. Since release 2.2 additional 10k pulldown resistor R94 has been added to project (https://github.com/sinara-hw/Thermostat/issues/94 ). This pulldown in combination with a 2k series resistor (R31 & R85) on the #SHDN line made a voltage divider and a high level voltage dropped below 3V. As a result, MAX1968 stopped turning on.

Removing the 10k pulldown makes the driver go out of shut-down and work.
@gkasprow

gkasprow, 2022-02-22:
right, but I'd add 100k pulldown to make sure the driver won't turn on when the CPU is not initialized.

gkasprow, 2022-02-22:
I would even replace R120 with a diode and add a pullup.
That would work both as a voltage shifter and wired-and gate.
To make sure the driver won't start when MCU is not initialized I'd add pulldowns at the STM PWM outputs
@maciejprzybysz

SingularitySurfer, 2022-02-22:
Ok, thanks.

SingularitySurfer, 2022-02-22:
Actually for developing now I'd only switch the pulldowns on #SHDN to 100k. I'll do more modifications as necessary.

jordens, 2022-02-22:
Yes. I meant for design changes.

SingularitySurfer, 2022-02-23:
Some more observations:
- No pulldown resistors on #SHDN leads to the following behavior:
    - If the CPU stops (I don't know what this means exactly, I simply stop execution by terminating the probe-rs process) the output current slowly creeps up to the maximum 3A.
    - The same happens when the board is powercycled with the CPU in reset. (There must be some glitch coming from the CPU that goes above the threshold voltage)
    - The PWM limit pins and the #SHDN pin creep to about 1.5V. (The PWM pins currently also have no pulldowns)
    - The DAC output goes to 0V which means maximum negative current for the TEC driver. 
- A 300K pulldown leads to a high level of 3.30V on the #SHDN pin, turning on the TEC at room temp. Stopping the CPU or powercycling wit CPU in reset leads to 0V and the TEC output is turned off.

gkasprow, 2023-03-06:
added level shifter and pulldown. This should solve the issue.
[image]

closed 2024-10-08 by filipswit