sinara-hw/Thermostat_EEM #59: header board for thermocouples

jordens, 2022-11-18:
Thermocouples are needed where dissipation or environment prevent the use of NTCs, RTDs, and diodes

* use the special connectors, which are type-dependent
* common reference junction closely coupled to ref junction thermistor on one of the channels
* excellent frontend amplifier (LTC1052?)
* design for very small temperature gradients across relevant frontend area
* careful design of the path between junctions and amplifier with an eye on undesired thermocouple voltages
* https://www.analog.com/media/en/technical-documentation/application-notes/an28f.pdf etc. and Keithley 2182A for ideas

gkasprow, 2022-11-18:
Maybe just add [Maxim integrated AFEs](https://www.maximintegrated.com/en/products/sensors/MAX31856.html) with I2C?

gkasprow, 2023-02-02:
We already have 2 front-end board options. Maybe we should make them as a separate project...

jordens, 2023-02-02:
Agreed

gkasprow, 2023-03-06:
Moving TC and Cryo breakout to separate PCB project